microsoft/Qcodes #6104: Alazar example (and driver) is broken

thangleiter, 2024-05-22:
Locating an Alazar board is broken due to an invalid identifier for a temporary instrument in `get_board_info()`.

### Steps to reproduce
1. Follow the [example notebook](https://microsoft.github.io/Qcodes/examples/driver_examples/Qcodes%20example%20with%20Alazar%20ATS9360.html), specifically,
    ```python
    AlazarTechATS9360.find_boards()
    ```

The problem lies in [`get_board_info()`](https://github.com/microsoft/Qcodes/blob/main/src/qcodes/instrument_drivers/AlazarTech/ATS.py#L106), which creates temporary instruments with the (unique) identifier `f'alazar_temp{time.perf_counter()}'`. However, `perf_counter()` returns a float, whose decimal point does not constitute a valid identifier:

```python
AlazarTechATS9360.find_boards()
Traceback (most recent call last):

  Cell In[5], line 1
    AlazarTechATS9360.find_boards()

  File ~\Documents\Qcodes\src\qcodes\instrument_drivers\AlazarTech\ATS.py:80 in find_boards
    boards.append(cls.get_board_info(api, system_id, board_id))

  File ~\Documents\Qcodes\src\qcodes\instrument_drivers\AlazarTech\ATS.py:106 in get_board_info
    board = cls(f'alazar_temp_{time.perf_counter()}',

  File ~\Documents\Qcodes\src\qcodes\instrument\instrument_meta.py:36 in __call__
    new_inst = super().__call__(*args, **kwargs)

  File ~\Documents\Qcodes\src\qcodes\instrument_drivers\AlazarTech\ATS9360.py:24 in __init__
    super().__init__(name, dll_path=dll_path, **kwargs)

  File ~\Documents\Qcodes\src\qcodes\instrument_drivers\AlazarTech\ATS.py:133 in __init__
    super().__init__(name, **kwargs)

  File ~\Documents\Qcodes\src\qcodes\instrument\instrument.py:68 in __init__
    super().__init__(name=name, metadata=metadata, label=label)

  File ~\Documents\Qcodes\src\qcodes\instrument\instrument_base.py:48 in __init__
    self._is_valid_identifier(self.full_name)

  File ~\Documents\Qcodes\src\qcodes\instrument\instrument_base.py:558 in _is_valid_identifier
    raise ValueError(f"{name} invalid instrument identifier")

ValueError: alazar_temp_504106.315358 invalid instrument identifier
```

The most straightforward solution seems to me to replace `perf_counter()` with `perf_counter_ns()`, which returns an integer instead of a float.

jenshnielsen, 2024-05-22:
Thanks @thangleiter I don't have access to an Alazar board at the moment but that seems like a reasonable fix.

thangleiter, 2024-05-22:
All right, I'll open a PR and make sure the example notebook runs through (I have an ATS9440 at hand).

PR #6106 (2024-05-23): Fix Alazar drivers
Fixes #6104.

The example now runs on my end (excepting trigger timeouts, which are likely to be device-sided).
closed 2024-05-23 by jenshnielsen