microsoft/Qcodes #6252: partial auto completion when coding

konosubakonoakua, 2024-07-16:
Bug report template:

###
For now, auto-completion partially works with LSP server (e.g. vscode), I can only see a little bit of method in the completion menu.
But, if I execute in ipython, then I can get the fulllists of methods

### Steps to reproduce
1. coding with vscode
2. create instance for instrument class e.g. `rigol_dg4000 : RigolDG4000`
3. type `rigol_dg4000.` to trigger the completion menu

### Expected behaviour
list all possible methods

### Actual behaviour
only partial of methods show up

### System
It would be helpful to provide such information:

**operating system**

If you are using a released version of qcodes (recommended):
**qcodes version** : 0.45.0

konosubakonoakua, 2024-07-16:
maybe it's because this lib designed with delegating, I cannot get full list of methods until it's running.

konosubakonoakua, 2024-07-16:
any workaround for this?

konosubakonoakua, 2024-07-16:
can we provide python stubs lib like pyqt?

konosubakonoakua, 2024-07-16:
or at least provide API docs in official docs here: https://microsoft.github.io/Qcodes/drivers_api/Rigol.html#qcodes.instrument_drivers.rigol.RigolDG4000

jenshnielsen, 2024-07-26:
@konosubakonoakua This is indeed because parameters / instrument modules / channel tuples are dynamically added to the instruments. In 0.46.0 nearly all parameters on instruments in qcodes were changed to be statically defined. It is the intension that the same will happen for instrument modules and channel tuples, however this is a significant amount of work so this may take some time. 

If you have instrument drivers outside qcodes that you would like to update qcodes ships with a tool to enable this. 
This tool is a CLI tool called `qcodes-refactor`

konosubakonoakua, 2024-07-29:
Ty :)