microsoft/Qcodes #6488: Keithley 2400 driver: source and sense range mixing

panasee, 2024-10-05:
```python
        self.add_parameter(
            "rangev",
            get_cmd="SENS:VOLT:RANG?",
            get_parser=float,
            set_cmd="SOUR:VOLT:RANG {:f}",
            label="Voltage range",
        )
```
Above is the range parameter for Keithley 2400. Here the get_cmd retrieves the sense range while the set_cmd tries to set the source range. From the manual of Keithley 2400, there can be two different ranges for sense and source. Are there special considerations for that?

jenshnielsen, 2024-10-07:
I don't have much context for this but the same is done for the current limit so I would guess it's probably intentional and an attempt to enable it to be used in both modes. That being said I don't think this is correct and I suggest that we add a parameter for both the sense and the set limit independently. @panasee would you be willing to open a pr to do that?

panasee, 2024-10-08:
I've further tested this with a 2401 meter. It seems the sense range is bound to the source range for the sourcing property. When I tried to set the sense range for the sourcing property, an error appeared at the meter's front panel (no error message from communication): "Invalid with source read-back on" (+823 error in the manual). This may be the reason for the original code writing.

I didn't find more details about the "source read-back" in the manual so I can't guarantee that it's a common feature for all 2400-series meters.

I think **the range parameters can be left as is for now** although it can be confusing.

The code I used is attached here:
```python
meter.mode("CURR")
meter.curr(1E-6)
meter.output(True)
meter.write("SENS:CURR:RANG 0.01")
meter.write("SOUR:CURR:RANG 0.001")
print(f"final source curr range: {meter.ask("SENS:CURR:RANG?")}")
print(f"final sense curr range: {meter.ask("SOUR:CURR:RANG?")}")
meter.output(False)

print("="*20)

meter.mode("VOLT")
meter.volt(1.5)
meter.output(True)
meter.write("SOUR:VOLT:RANG 10")
meter.write("SENS:VOLT:RANG 1")
print(f"final source volt range: {meter.ask("SENS:VOLT:RANG?")}")
print(f"final sense volt range: {meter.ask("SOUR:VOLT:RANG?")}")
meter.output(False)
```
The output is:
```
final source curr range: 1.050000E-03
final sense curr range: 1.050000E-03
====================
final source volt range: 21.00
final sense volt range: 21.00
```
Apparently, the "SENS:CURR(VOLT):RANG" caused no effect here.

panasee, 2024-10-08:
Sometimes, the meter will give some error messages that only show up on its front panel. Is there a way to capture them in python? This could be very useful for debugging.