microsoft/Qcodes #6975: ConnectionResetError when Controlling an Oxford Triton Fridge via QCoDeS Interface

johnnyzhaophysics, 2025-03-18:
I encountered a ConnectionResetError while controlling an Oxford Instruments Triton fridge using the QCoDeS interface. The traceback for the error is attached as below.

[connectionerrormessage.docx](https://github.com/user-attachments/files/19317519/connectionerrormessage.docx)

I have tried increasing the timeout before setting the field, adding a delay between commands but the issue persists.

Question:
Has anyone experienced similar issues when controlling the Triton magnet power supply via Qcodes? 
Are there known fixes or best practices to prevent the connection from being reset?
Is there a recommended way to handle this in QCoDeS (e.g., reconnecting if the connection is lost)?

Any insights would be greatly appreciated! Thanks in advance.

johnnyzhaophysics, 2025-03-18:
Here is my measurement code:

### Measurement Code:
```python
from qcodes.instrument_drivers.oxford.triton import Triton as qcodes_Triton
triton = qcodes_Triton(name='Oxford_T1', address='172.16.39.21', port=33576, timeout=100000, persistent=False)

magnet_field_x = Parameter('magnet_field_x', unit='T', label='$B_x$', set_cmd=triton.Bx, get_cmd=triton.Bx)
magnet_field_y = Parameter('magnet_field_y', unit='T', label='$B_y$', set_cmd=triton.By, get_cmd=triton.By)
magnet_field_z = Parameter('magnet_field_z', unit='T', label='$B_z$', set_cmd=triton.Bz, get_cmd=triton.Bz)

check_and_add_component(magnet_field_x)
check_and_add_component(magnet_field_y)
check_and_add_component(magnet_field_z)

def set_polar_r(r_val):
    pi = np.pi
    theta = 0
    phi = 90  # 90 degrees is π/2 radians
    
    # Convert angles from degrees to radians
    phi_rad = phi * pi / 180
    theta_rad = theta * pi / 180
    
    # Calculate the x, y, z components of the magnetic field
    x_target = r_val * np.sin(phi_rad) * np.cos(theta_rad)
    y_target = r_val * np.sin(phi_rad) * np.sin(theta_rad)
    z_target = r_val * np.cos(phi_rad)
    
    # Set the magnet field components
    magnet_field_x(x_target)
    print(f'set x_target to {x_target} T')
    x_current = triton.Bx()
    x_togo = np.abs(x_target - x_current)
    x_time = x_togo * 60 / 0.05
    sleep(x_time + 30)
    
    magnet_field_y(y_target)
    print(f'set y_target to {y_target} T')
    y_current = triton.By()
    y_togo = np.abs(y_target - y_current)
    y_time = y_togo * 60 / 0.05
    sleep(y_time + 30)
    
    magnet_field_z(z_target)
    print(f'set z_target to {z_target} T')
    z_current = triton.Bz()
    z_togo = np.abs(z_target - z_current)
    z_time = z_togo * 60 / 0.05
    sleep(z_time + 30)

magnet_field_r = Parameter('R', unit='T', label='R', set_cmd=set_polar_r, get_cmd=None)
triton.magnet_sweeprate(0.05)
```

jenshnielsen, 2025-03-21:
@johnnyzhaophysics I have not used the Triton driver for a long time. As I remember it there was a number of issues with the Triton interface provided and we usually used the Mercury driver directly